3arthur6/BluetoothLibraryPatcher #7: Samsung Galaxy A80 Android 11 not supported

TheSingular, 2022-08-07:
Title explains it well I believe. I have repacked the .tar into .zip and uploaded it here, link is below.

[BluetoothLibPatcher-files.zip](https://github.com/3arthur6/BluetoothLibraryPatcher/files/9277702/BluetoothLibPatcher-files.zip)

3arthur6, 2022-08-09:
Hey
I didn't found why it fails.
Could you try again with the latest release and post the new tar file?

TheSingular, 2022-08-09:
> Hey I didn't found why it fails. Could you try again with the latest release and post the new tar file?

TheSingular, 2022-08-09:
> > Hey I didn't found why it fails. Could you try again with the latest release and post the new tar file?

Sure, did it just now and it failed again. 
[image]

New tar below.
[BluetoothLibPatcher-files.zip](https://github.com/3arthur6/BluetoothLibraryPatcher/files/9294964/BluetoothLibPatcher-files.zip)

Also accidentally closed the issue just now, reopened it so please ignore that.

TheSingular, 2022-08-09:
Hey I just tried to install the v2.3.1 version from the old repo and it worked somehow. I don't know why. At least I won't lose my pairings again.

3arthur6, 2022-08-09:
Could you give it one last try?

TheSingular, 2022-08-09:
Yeah the lastest one worked.

closed 2022-08-09 by TheSingular