3arthur6/BluetoothLibraryPatcher #12: S22 Ultra

wiredroid, 2022-08-12:
Hey, Thanks for this module. I connect to my Bluetooth devices ok but my watch 4 keeps resetting and saying to pair with your new device you have to reset watch. My issue is I'm running Magisk only device is still encrypted with stock recovery and stock kernel as I don't want to run a passive kernel. Is there anything I can do?

wiredroid, 2022-08-12:
@huskydg sent me this. Not sure where to put it but maybe you could update the library.

"If magisk modules mount too late, you can use early-init mount"

for i in `grep -lr 'security.wsm' /vendor/etc/vintf`; do
    if [ ! -z $i ]; then
         mkdir -p "$(magisk --path)/.magisk/mirror/early-mount/system/vendor/vintf/$i"
         cp -af "$i" "$(magisk --path)/.magisk/mirror/early-mount/system/vendor/vintf/$i" 
         sed -i "$((`awk '/security.wsm/ {print FNR}' $i`-1)),/<\/hal>/d" "$(magisk --path)/.magisk/mirror/early-mount/system/vendor/vintf/$i";
    fi

HuskyDG, 2022-08-12:
Early mount is Magisk Delta specific feature.

```
for i in `grep -lr 'security.wsm' /vendor/etc/vintf`; do
    rm -rf "$(magisk --path)/.magisk/mirror/early-mount/system/vendor/vintf/$i" 
    if [ ! -z $i ]; then
         mkdir -p "$(magisk --path)/.magisk/mirror/early-mount/system/vendor/vintf/${i%/*}"
         cp -af "$i" "$(magisk --path)/.magisk/mirror/early-mount/system/vendor/vintf/$i" 
         sed -i "$((`awk '/security.wsm/ {print FNR}' $i`-1)),/<\/hal>/d" "$(magisk --path)/.magisk/mirror/early-mount/system/vendor/vintf/$i";
    fi
done
```

3arthur6, 2022-08-13:
Hi!
Sadly this isn't relevant with official release. Any plan for a pull request of this feature?
Is MAGISK_VER variable always ended with "delta"? I need a way to detect your magisk fork.

wiredroid, 2022-08-13:
Detect the existence of $(magisk --path)/.magisk/mirror/early-mount

HuskyDG, 2022-08-13:
> Hi! Sadly this isn't relevant with official release. Any plan for a pull request of this feature? Is MAGISK_VER variable always ended with "delta"? I need a way to detect your magisk fork.

99,9% the author of Magisk will not want to add this feature.

3arthur6, 2022-08-13:
I pushed a commit to handle this. Thanks for the suggestion.

closed 2022-08-13 by 3arthur6