3arthur6/BluetoothLibraryPatcher #13: SM-G973F

aliafshany, 2022-08-12:
hello there 
i tried to flash the module via Magisk, but it failed and led me here.

this is my logs.
please investigate this

[BluetoothLibPatcher-files.tar.zip](https://github.com/3arthur6/BluetoothLibraryPatcher/files/9318030/BluetoothLibPatcher-files.tar.zip)
.

3arthur6, 2022-08-13:
Hey
Humm not sure what happened.
Try again or download the latest release [here](https://github.com/3arthur6/BluetoothLibraryPatcher/releases/download/v2.4.2/BluetoothLibraryPatcher_magisk_2.4.2.zip) and install it in magisk from local storage.

aliafshany, 2022-08-13:
i installed the latest version, and it's working
initially, i flashed it from "Fox's Mmm", maybe there was an old version of it.

btw thank you

closed 2022-08-13 by 3arthur6