3arthur6/BluetoothLibraryPatcher #15: TWRP Patcher Fail - Error 2

stylemessiah, 2022-08-17:
Just trying on latest ROM on Samsung s20+

TWRP 3.6.2-11.0

Redownloaded

Fails consistently with log content:

 - Bluetooth Library Patcher v2.4.2 -

- Mounting system & vendor
mount: '/system' not in fstab
- Extracting tools
**/tmp/bluetoothlibpatcher/bin/hexpatch.sh: line 154: syntax error near unexpected token `fi'
/tmp/bluetoothlibpatcher/bin/hexpatch.sh: line 154: `    fi'**
Updater process ended with ERROR: 2
I:Install took 0 second(s).
Error installing zip file '/sdcard/BluetoothLibraryPatcher_twrp_2.4.2.zip'
Updating partition details...

3arthur6, 2022-08-18:
Hey
Sorry there was an issue with 2.4.2. Try again with 2.4.3

stylemessiah, 2022-08-18:
Cheers Arthur :)

closed 2022-08-18 by stylemessiah