3arthur6/BluetoothLibraryPatcher #23: Slow script execution during boot on Galaxy Note 9 running Android 10

brusyev, 2022-09-04:
It appears that during boot the latest version of BluetoothLibraryPatcher is running the module script noticeably slower on Galaxy Note 9 (Android 10) and latest version of TWRP Crownlite. Not sure if this is TWRP related or something else is causing it to run slower.

brusyev, 2022-09-07:
This would be the latest version of magisk module. Upon the reboot it takes a few minutes for the script to kick in to temporarily disable bluetooth, enable airplane mode.

3arthur6, 2022-09-08:
What you say is not clear.
What is wrong? The patch works right so?

brusyev, 2022-09-10:
The patch works great. When the phone is restarted it looks like the patch runs a script which temporarily disabled bluetooth and then enables it again. This process started taking longer for some reason. Otherwise the patch works great. Thank you for all the work on this.

3arthur6, 2022-09-10:
Humm no this clearly doesn't disable nor enable bluetooth at any moments.
Not sure what is your issue but it's not related to this.

closed 2022-09-10 by 3arthur6