3arthur6/BluetoothLibraryPatcher #25: Module flashed but not working.

SharkBoy69-CS, 2022-11-04:
Hey, I flashed the module but everytime I reboot my phone I'm still losing Bluetooth paired devices. How could I fix it ? Did I make something wrong ?

3arthur6, 2022-11-08:
Hey
Humm did the installation says it was successfully applied?
Which phone? Which firmware version?

SharkBoy69-CS, 2022-11-08:
> Hey
> Humm did the installation says it was successfully applied?
> Which phone? Which firmware version?

Yes, it said successful but I lost my paired device after each reboot
I'm on a sm-975f and oneui 4.1

SharkBoy69-CS, 2022-11-08:
I'll try to reset network settings then

SharkBoy69-CS, 2022-11-26:
> Hum it should be enough.
> Only one device?
> Did you try with other devices? And/or reset network settings?
> 

I tried and it still doesn't work... I also tried your last update but it's not working too. Could it be because I'm using Riru and not Zygisk ? Or because I'm using Magisk Delta with SuList ?

3arthur6, 2022-11-26:
Could you upload here your /data/misc/bluedroiddump/subBuffer.log file please?

SharkBoy69-CS, 2022-11-26:
Here are the logs (I connected my earphones since the last time I rebooted, tell me if I need to reboot my phone without pairing back my earphones)

https://www.mediafire.com/file/mgkz2yyb4wtyfyq/subBuffer.log/file

3arthur6, 2022-11-26:
Hum there is something wrong yes.
Could you also send a copy of /data/adb/modules/BluetoothLibraryPatcher folder?

SharkBoy69-CS, 2022-11-26:
Yes sure :D

https://www.mediafire.com/file/948mqduym7d2k7k/BluetoothLibraryPatcher.zip/file

3arthur6, 2022-11-26:
The module seems fine.
Are you sure you it is activated in magisk app's modules tab?
Otherwise there is an issue with your config. Maybe something with magisk delta or riru as you stated. You should try with a common magisk config.

SharkBoy69-CS, 2022-11-26:
I suppose it's due to SuList in Magisk Delta because it is at a beta statement and some modules aren't working with it

Thanks for your help ^^
And thanks for your module it was very useful when I wasn't using SuList 🙏

closed 2022-11-27 by 3arthur6