3arthur6/BluetoothLibraryPatcher #27: S22 A13

bklynali, 2022-11-11:
I see you pulled the android 13 off but on my s22 ultra on android 13 after restart I have to readd the Bluetooth devices.

3arthur6, 2022-11-13:
Hey
It's not needed anymore on A13.
You probably removed vaultkeeper service, right?

freeza-inc, 2022-11-24:
I haven't touched vaultkeeper service and it loses pairings on reboot for me too

bklynali, 2022-11-24:
The only thing you should replace is vendor/etc/vintf/wsm_manifest.xml nothing else that worked for me.

freeza-inc, 2022-11-25:
> The only thing you should replace is vendor/etc/vintf/wsm_manifest.xml nothing else that worked for me.

I don't have a wsm_manifest in vintf. S20 ultra here. Several people on the s20 series have confirmed Bluetooth pairings being lost on reboot and toggling Bluetooth on and off 🤔🤔

bklynali, 2022-11-25:
I have Note 20 ultra I did the following no issues.
/vendor/etc/vintf/manifest << folder all the way at the bottom. is wsm_manifest.xml

open that file and replace everything with.

<manifest version="1.0" type="device">
</manifest>

Remove the following from /vendor/etc/vintf/manifest.xml

3arthur6, 2022-11-25:
> I haven't touched vaultkeeper service and it loses pairings on reboot for me too

So are you loosing all paired devices or only gear watch?
And did you flashed multidisabler or something like this?

bklynali, 2022-11-25:
No issues at all with bluetooth and I just updated it to android 13 SM-N986U

freeza-inc, 2022-11-25:
> > I haven't touched vaultkeeper service and it loses pairings on reboot for me too
> 
> So are you loosing all paired devices or only gear watch? And did you flashed multidisabler or something like this?

all paired devices. loses them instantly when toggling bluetooth on and off and/or rebooting. @afaneh92 and several others have confirmed it happens on their s20 series devices as well. patched and unpatched manifest.xml doesn't matter. in fact, patching manifest doesn't seem to have any effect at all on anything bluetooth related. bluetooth.apk has also been moved to com.android.btservices.apex on A13

freeza-inc, 2022-11-25:
> I have Note 20 ultra I did the following no issues. /vendor/etc/vintf/manifest << folder all the way at the bottom. is wsm_manifest.xml
> 
> open that file and replace everything with.
> 
> Remove the following from /vendor/etc/vintf/manifest.xml vendor.samsung.hardware.security.wsm

[image]

3arthur6, 2022-11-25:
> all paired devices. loses them instantly when toggling bluetooth on and off and/or rebooting. @afaneh92 and several others have confirmed it happens on their s20 series devices as well. patched and [...]

Could you upload a copy of your /data/misc/bluedroiddump/subBuffer.log?

I get some feedback about that. It seems some have the issue and others not.
Actually I successfully patched com.android.btservices apex and it run pretty well, in my S20 at least.
But it's a signed stuff like system apks, so you can easily fall into bootloop if you modify it.
And in addition it's loaded before magisk starts, so not module compatible.

freeza-inc, 2022-11-25:
> Could you upload a copy of your /data/misc/bluedroiddump/subBuffer.log?
>
> I get some feedback about that. It seems some have the issue and others not. Actually I successfully patched [...]

sure!

awesome to hear you've already got a fix for it too. minor inconvenience about magisk but hey, a fix is a fix :)
[subBuffer.log](https://github.com/3arthur6/BluetoothLibraryPatcher/files/10095025/subBuffer.log)

3arthur6, 2022-11-25:
> 
> sure!
> 
> awesome to hear you've already got a fix for it too. minor inconvenience about magisk but hey, a fix is a fix :) [subBuffer.log](https://github.com/3arthur6/BluetoothLibraryPatcher/files/10095025/subBuffer.log)

Ok it's clearly related to vaultkeeper in your log. Not sure why..
Are you running on custom kernel and/or did you debloated some system apps?

If you want to try the fix, at your own risk.
Go to recovery, make a backup of super image and/or com.android.btservices.apex.
Your recovery need busybox support.

Then run these commands throught terminal or run it as script:

`cp -f /system_root/system/apex/com.android.btservices.apex /tmp
busybox xxd -p /tmp/com.android.btservices.apex|tr -d ' \n'|sed 's/6804003528008052/2a00001428008052/'|busybox xxd -pr > /system_root/system/apex/com.android.btservices.apex
NOW=$(date) && date "`stat -c %z /system_root/system/apex/com.android.runtime.apex|cut -b -16`" > /dev/null && touch /system_root/system/apex/com.android.btservices.apex && date "$NOW" > /dev/null
rm -f /tmp/com.android.btservices.apex`

freeza-inc, 2022-11-25:
> > sure!
> > awesome to hear you've already got a fix for it too. minor inconvenience about magisk but hey, a fix is a fix :) [subBuffer.log](https://github.com/3arthur6/BluetoothLibraryPatcher/files/10095025/subBuffer.log)
> 
> Ok it's clearly related to vaultkeeper in your log. Not sure why.. Are you running on custom kernel and/or did you debloated some system apps?
> 
> If you want to try the fix, at your own risk. Go to recovery, make a backup of super image and/or com.android.btservices.apex. Your recovery need busybox support.
> 
> Then run these commands throught terminal or run it as script:
> 
> `mount -w /system_root; cp -f /system_root/system/apex/com.android.btservices.apex /tmp; busybox xxd -p /tmp/com.android.btservices.apex|tr -d ' \n'|sed 's/6804003528008052/2a00001428008052/'|busybox xxd -pr > /system_root/system/apex/com.android.btservices.apex; NOW=$(date) && date "$(stat -c %z /system_root/system/apex/com.android.runtime.apex|cut -b -16)" > /dev/null && touch /system_root/system/apex/com.android.btservices.apex && date "$NOW" > /dev/null; rm -f /tmp/com.android.btservices.apex; umount /system_root; `

Incredible. Worked perfectly. Thank you SO much!

bklynali, 2022-11-26:
> Ok it's clearly related to vaultkeeper in your log. Not sure why.. Are you running on custom kernel and/or did you debloated some system apps?
>
> If you want to try the fix, at your own risk. Go to [...]

Can this be done using magisk hexpatch?

bklynali, 2022-11-26:
> Could you upload a copy of your /data/misc/bluedroiddump/subBuffer.log?
>
> I get some feedback about that. It seems some have the issue and others not. Actually I successfully patched [...]

If this issue comes up in S22 you can't patch it in twrp since since everything is read only. Is there a way to patch on the computer and load it into rom

jrkruse, 2022-11-26:
> Ok it's clearly related to vaultkeeper in your log. Not sure why.. Are you running on custom kernel and/or did you debloated some system apps?
>
> If you want to try the fix, at your own risk. Go to [...]

This works but the only issue is if you wipe data after the com.android.btservices.apex is patched phone will not boot

3arthur6, 2022-11-26:
> 
> This works but the only issue is if you wipe data after the com.android.btservices.apex is patched phone will not boot on s20+ with a13

Yes as explained above it's because of some signature check.
On clean install, after an update and most probably in other situations, you need to be sure the patch isn't applied, boot one first time and then reboot to recovery to apply the patch.

3arthur6, 2022-11-26:
> Can this be done using magisk hexpatch?

Sure yep!

> If this issue comes up in S22 you can't patch it in twrp since since everything is read only. Is there a way to patch on the computer and load it into rom

Is there no way to make it writable through recovery at this point?

bklynali, 2022-11-26:
> > Can this be done using magisk hexpatch?
> 
> Sure yep!
> 
> > If this issue comes up in S22 you can't patch it in twrp since since everything is read only. Is there a way to patch on the computer and load it into rom
> 
> Is there no way to make it writable through recovery at this point?

Nope everything is R/O you can try to make system RW but its a pain including vendor. I did hex patch it put in the rom and flashed it. It just bootloops, you are correct it is checking for signature but need to find out what type of. on A13 I did disable V2 signature to modified systems apps.

3arthur6, 2022-11-27:
Ok guys I finally fund a way to turn it into magisk module.
Let me just rewrite the module to support it.
This will fix bootloops issue.

Conan179, 2022-11-27:
With the new version, bt is broken on my fold4, it can no longer be activated, the switch immediately jumps back to off.

3arthur6, 2022-11-27:
> With the new version, bt is broken on my fold4, it can no longer be activated, the switch immediately jumps back to off.

On Android 13?
Could you upload here your /system/apex/com.android.btservices.apex file please?

Conan179, 2022-11-27:
Yes A13.
No problem. 
[btservices.zip](https://github.com/3arthur6/BluetoothLibraryPatcher/files/10098384/btservices.zip)

3arthur6, 2022-11-27:
> Yes A13. No problem. [btservices.zip](https://github.com/3arthur6/BluetoothLibraryPatcher/files/10098384/btservices.zip)

Hum strange.. your apex file is similar to S20 one, it should work..

Could you install again the module and upload /data/adb/modules_update/BluetoothLibraryPatcher folder?
No need to reboot and you can remove the folder or disable the module just after.

Conan179, 2022-11-27:
Yes, no problem
[BluetoothLibraryPatcher.zip](https://github.com/3arthur6/BluetoothLibraryPatcher/files/10098454/BluetoothLibraryPatcher.zip)

3arthur6, 2022-11-27:
> Yes, no problem [BluetoothLibraryPatcher.zip](https://github.com/3arthur6/BluetoothLibraryPatcher/files/10098454/BluetoothLibraryPatcher.zip)

Huh.. All seems fine in the module folder too. No clue why it doesn't work on Fold 4 but it does on S20..
I need to take a look on Fold firmware.

Conan179, 2022-11-27:
I'm very happy to do it for you, you just have to tell me how to do it.

3arthur6, 2022-11-27:
> I'm very happy to do it for you, you just have to tell me how to do it.

Install the module, reboot, try to activate bluetooth.
Then take a logcat dump (look on Google if you never did that) and upload it.

Conan179, 2022-11-27:
Ok
[logcat_11-27-2022_15-43-13.txt](https://github.com/3arthur6/BluetoothLibraryPatcher/files/10098531/logcat_11-27-2022_15-43-13.txt)

3arthur6, 2022-11-27:
> Ok [logcat_11-27-2022_15-43-13.txt](https://github.com/3arthur6/BluetoothLibraryPatcher/files/10098531/logcat_11-27-2022_15-43-13.txt)

This looks not like a logcat
Again try to activate bugged bluetooth.
Then from termux or terminal app, execute these commands
`su
logcat -d > /sdcard/logcat.log`
And upload logcat.log from your internal storage

Conan179, 2022-11-27:
[logcat.log](https://github.com/3arthur6/BluetoothLibraryPatcher/files/10098569/logcat.log)

3arthur6, 2022-11-27:
> > This looks not like a logcat
> 
> but was created with https://play.google.com/store/apps/details?id=com.dp.logcatapp&hl=de
> 
> > Again try to activate bugged bluetooth. Then from termux or terminal app, execute these commands `su logcat -d > /sdcard/logcat.log` And upload logcat.log from your internal storage
> 
> ok [logcat.log](https://github.com/3arthur6/BluetoothLibraryPatcher/files/10098569/logcat.log)

Thanks for the log.

Could you try with this one?
[BluetoothLibraryPatcher_magisk_2.6_test.zip](https://github.com/3arthur6/BluetoothLibraryPatcher/files/10098758/BluetoothLibraryPatcher_magisk_2.6_test.zip)

Conan179, 2022-11-27:
[logcat2.log](https://github.com/3arthur6/BluetoothLibraryPatcher/files/10098777/logcat2.log)

bklynali, 2022-11-27:
> yes sure. It doesn't work, but something has changed, now I see the message that Bluethotto force is being closed. I didn't even see the message with the other version because it immediately disappeared. [logcat2.log](https://github.com/3arthur6/BluetoothLibraryPatcher/files/10098777/logcat2.log)

Did you deodex your rom? because I did that and I was having these issues.

3arthur6, 2022-11-27:
> yes sure. It doesn't work, but something has changed, now I see the message that Bluethotto force is being closed. I didn't even see the message with the other version because it immediately disappeared. [logcat2.log](https://github.com/3arthur6/BluetoothLibraryPatcher/files/10098777/logcat2.log)

Huh.. You could these one but at this point i'm trying some kind of random changes..
[BluetoothLibraryPatcher_magisk_2.6_test_w8_1.zip](https://github.com/3arthur6/BluetoothLibraryPatcher/files/10098812/BluetoothLibraryPatcher_magisk_2.6_test_w8_1.zip)
[BluetoothLibraryPatcher_magisk_2.6_test_w8_0.zip](https://github.com/3arthur6/BluetoothLibraryPatcher/files/10098810/BluetoothLibraryPatcher_magisk_2.6_test_w8_0.zip)
[BluetoothLibraryPatcher_magisk_2.6_test_w25w19.zip](https://github.com/3arthur6/BluetoothLibraryPatcher/files/10098813/BluetoothLibraryPatcher_magisk_2.6_test_w25w19.zip)

Conan179, 2022-11-27:
No, i have stock rom with root by patched ap file.

Conan179, 2022-11-27:
With BluetoothLibraryPatcher_magisk_2.6_test_w8_0.zip it worked immediately and reset my watch. crap samsung health lost data from the last few days where the watch was not connected.

3arthur6, 2022-11-27:
> With BluetoothLibraryPatcher_magisk_2.6_test_w8_0.zip it worked immediately and reset my watch. crap! samsung health lost data from the last few days where the watch was not connected.

So it works? Not sure to understand?

Conan179, 2022-11-27:
I'm sorry, i speak german and use google translate.

Yes, with BluetoothLibraryPatcher_magisk_2.6_test_w8_0.zip its working fine!

3arthur6, 2022-11-27:
> I'm sorry, i speak german and use google translate.
> 
> Yes, with BluetoothLibraryPatcher_magisk_2.6_test_w8_0.zip its working fine!

Nice to hear!
Thanks for the help I will update the module.

jrkruse, 2022-11-27:
> > I'm sorry, i speak german and use google translate.
> > Yes, with BluetoothLibraryPatcher_magisk_2.6_test_w8_0.zip its working fine!
> 
BluetoothLibraryPatcher_magisk_2.6_test_w8_0.zip doesnt seem to work on SM-G986U but the BluetoothLibraryPatcher_magisk_2.6_test_w8_1.zip does

3arthur6, 2022-11-27:
> > > I'm sorry, i speak german and use google translate.
> > > Yes, with BluetoothLibraryPatcher_magisk_2.6_test_w8_0.zip its working fine!
> 
> BluetoothLibraryPatcher_magisk_2.6_test_w8_0.zip doesnt seem to work on SM-G986U but the BluetoothLibraryPatcher_magisk_2.6_test_w8_1.zip does

Thanks!
I just figured out same thing on my device.
@Conan179 same for you? Did you mean w8_1 worked?

jrkruse, 2022-11-27:
> Thanks! I just figured out same thing on my device. @Conan179 same for you? Did you mean w8_1 worked?

actually all the 2.6 zips worked except the w8_0 one

Conan179, 2022-11-27:
No w8_1and w25w19 dosnt work on my Fold4

jrkruse, 2022-11-27:
> No w8_1and w25w19 dosnt work on my Fold4

yes what works on your fold4 doesnt work on the s20 is what I am saying

3arthur6, 2022-11-27:
> actually all the 2.6 zips worked except the w8_0 one

So even the first one from github release works, right?
I will wait for more feedback before pushing update.

jrkruse, 2022-11-27:
> > actually all the 2.6 zips worked except the w8_0 one
> 
> So even the first one from github release works, right? I will wait for more feedback before pushing update.

Yes it worked as well as did all the test ones except this one BluetoothLibraryPatcher_magisk_2.6_test_w8_0.zip

3arthur6, 2022-11-27:
> No w8_1and w25w19 dosnt work on my Fold4

And does w8_0 fixed well the pairing issue?

3arthur6, 2022-11-27:
> yes

So to be sure, without the module, after a reboot, you can not recover any saved paired devices but with the module they stay saved, right?
And this apply not only for gear watch but any bluetooth paired devices.

Conan179, 2022-11-27:
I refer to https://github.com/3arthur6/BluetoothLibraryPatcher/issues/24
That's been the same problem.
My Galaxy Bus Live was always connected even after a reboot.
My SPen Pro is always connected, the only problem I had was that I couldn't connect my watches (watch3 and Gear s2).

bklynali, 2022-11-27:
> Huh.. You could these one but at this point i'm trying some kind of random changes.. [...]

---------------
[BluetoothLibraryPatcher_magisk_2.6_test_w8_1.zip] Works well on Note 20.

3arthur6, 2022-11-27:
> [BluetoothLibraryPatcher_magisk_2.6_test_w8_1.zip] Works well on Note 20.
> 
> /BluetoothLibraryPatcher_magisk_2.6_test_w8_0.zip) < Bluetooth keeps disconnecting and connecting.

And what about your S22?

bklynali, 2022-11-28:
> > [BluetoothLibraryPatcher_magisk_2.6_test_w8_1.zip] Works well on Note 20.
> > /BluetoothLibraryPatcher_magisk_2.6_test_w8_0.zip) < Bluetooth keeps disconnecting and connecting.
> 
> And what about your S22?

Right now s22 doesn't require patch but I flash it didn't notice any difference. Maybe next update when Samsung changes vendor and kernel to android 13 we'll know for sure

bklynali, 2022-11-30:
> `mount -w /system_root; cp -f /system_root/system/apex/com.android.btservices.apex /tmp; busybox xxd -p /tmp/com.android.btservices.apex|tr -d ' \n'|sed [...]

Quick question when we change 6804003528008052 to this 2a00001428008052 what does it change?

3arthur6, 2022-11-30:
> Quick question when we change 6804003528008052 to this 2a00001428008052 what does it change?

It's arm instructions converted to hexadecimal.
In quick it bypass the vaultkeeper check by jumping at the end of the checking fonction.
Use a binary disassembler and compare to to see in detail.

bklynali, 2022-12-03:
> > Quick question when we change 6804003528008052 to this 2a00001428008052 what does it change?
> 
> It's arm instructions converted to hexadecimal. In quick it bypass the vaultkeeper check by jumping at the end of the checking fonction. Use a binary disassembler and compare to to see in detail.
------------------------------------------------------------------------------
You can close this issue 2.6.1 works on S22 I just tired it I removed vaultkeeper and stuff and flashed the zip and it worked without any issue

3arthur6, 2022-12-03:
> You can close this issue 2.6.1 works on S22 I just tired it I removed vaultkeeper and stuff and flashed the zip and it worked without any issue

Thanks for the feedback

closed 2022-12-03 by 3arthur6