3arthur6/BluetoothLibraryPatcher #31: A52s Bluetooth broken

ThePutois, 2022-12-27:
Hello, I installed this twrp zip hoping it would allow me to pair m'y galaxy watch 5 pro to m'y rooted a52s (stock rom), the flashing worked but bluetooth is now greyed out and unusable, I saw That the original file is backed up during patching, how to restore it please?

3arthur6, 2022-12-28:
Hey
Sorry for that. So on Android 13 right?
On A13 replace /system_root/system/lib64/libbluetooth_jni.so by /system_root/system/lib64/libbluetooth_jni.so.bak from twrp.
Otherwise in A12 do the same stuff but with libbluetooth.so

ThePutois, 2022-12-28:
Hi, thanks for your reply, I did what you wrote but I still have the issue, do I need to refresh something more than restarting the device ? Thanks

3arthur6, 2022-12-28:
If you are on Android 13 totally remove the file.

3arthur6, 2023-02-03:
No response, so closing.

closed 2023-02-03 by 3arthur6