3arthur6/BluetoothLibraryPatcher #32: Samsung Galaxy A13 library not found

Exynos-nibba, 2023-01-02:
**Device: Samsung Galaxy A13 
OS: Android 13 (OneUI 5 core)**

Module installation fails on _Android 13_ with a 

> library not found error

Device is **MTK based** so im guessing it has differently named or labeled libs? and no _lib64_ folder.
note: There is a _libbluetooth_mtk_ lib in the vendors folder

Magisk Installation log & a complete list of libs from the system and vendor folders is attached below and im guessing you would need a lot more than this, feel free to ask more logs, etc to debug the issue, thanks. 

[magisk_install_log_2023-01-02T20.07.50.log](https://github.com/3arthur6/BluetoothLibraryPatcher/files/10332608/magisk_install_log_2023-01-02T20.07.50.log)

[system-libs.txt](https://github.com/3arthur6/BluetoothLibraryPatcher/files/10332619/system-lib.txt)

[vendor-libs.txt](https://github.com/3arthur6/BluetoothLibraryPatcher/files/10332620/vendor-libs.txt)

3arthur6, 2023-01-04:
Most users don't need this module anymore on android 13, are you still losing bluetooth pairings?
Anyway I take a look on the Galaxy A13 firmware and the latest release (2.6.2) should handle it.

Exynos-nibba, 2023-01-04:
> Most users don't need this module anymore on android 13, are you still losing bluetooth pairings? Anyway I take a look on the Galaxy A13 firmware and the latest release (2.6.2) should handle it.

Yep, the device still loses pairings even if i turn the bluetooth off and on after a few mins, 2.6.2 seems to have flashed successfully, will close the issue if the pairings stay after this, Thanks!

Exynos-nibba, 2023-01-04:
Tested it, works perfectly fine.

closed 2023-01-04 by Exynos-nibba