3arthur6/BluetoothLibraryPatcher #45: Still drop pairing on Zflip4

XDZemd, 2023-09-23:


XDZemd, 2023-09-23:
here is the log:

- Copying zip to temp directory
- Installing BluetoothLibraryPatcher_magisk_2.6.4.zip
- Device is system-as-root
Archive:  /data/user/0/itioag.j/cache/flash/install.zip
  inflating: module.prop
***************************
 Bluetooth Library Patcher 
 by 3arthur6 
***************************
*******************
 Powered by Magisk 
*******************
Archive:  /data/user/0/itioag.j/cache/flash/install.zip
  inflating: customize.sh
- Extracting module files
Archive:  /data/user/0/itioag.j/cache/flash/install.zip
  inflating: 7z.tar.xz
  inflating: README.md
  inflating: SUPPORT.md
  inflating: bash.tar.xz
  inflating: busybox.tar.xz
  inflating: customize.sh
  inflating: hexpatch.sh
  inflating: latestchange.log
  inflating: module.prop
  inflating: service.sh
  inflating: update.json
- Magisk Manager installation
- Searching for relevant hex byte sequence
cp: can't stat '/dev/lxnjmokxiwmgx/.magisk/mirror/system/../apex/com.android.btservices/lib64/libbluetooth_jni.so': No such file or directory
xxd: /dev/tmp/system/lib64/libbluetooth_jni.so: No such file or directory
xxd: /dev/tmp/system/lib64/libbluetooth_jni.so: No such file or directory
BusyBox v1.34.1-Magisk (2022-08-19 10:51:18 PDT) multi-call binary.

Usage: grep [-HhnlLoqvsrRiwFE] [-m N] [-A|B|C N] { PATTERN | -e PATTERN... | -f FILE... } [FILE]...

Search for PATTERN in FILEs (or stdin)

	-H	Add 'filename:' prefix
	-h	Do not add 'filename:' prefix
	-n	Add 'line_no:' prefix
	-l	Show only names of files that match
	-L	Show only names of files that don't match
	-c	Show only count of matching lines
	-o	Show only the matching part of line
	-q	Quiet. Return 0 if PATTERN is found, 1 otherwise
	-v	Select non-matching lines
	-s	Suppress open and read errors
	-r	Recurse
	-R	Recurse and dereference symlinks
	-i	Ignore case
	-w	Match whole words only
	-x	Match whole lines only
	-F	PATTERN is a literal (not regexp)
	-E	PATTERN is an extended regexp
	-m N	Match up to N times per file
	-A N	Print N lines of trailing context
	-B N	Print N lines of leading context
	-C N	Same as '-A N -B N'
	-e PTRN	Pattern to match
	-f FILE	Read pattern from file
xxd: /dev/tmp/system/lib64/libbluetooth_jni.so: No such file or directory
- Applying patch
sh: already: unknown operand
cp: can't stat '/dev/tmp/system/lib64/libbluetooth_jni.so': No such file or directory
- Patch failed!
 
- Opening support webpage in 10 seconds

! Installation failed

3arthur6, 2023-09-27:
Go back to [v2.6.3](https://github.com/3arthur6/BluetoothLibraryPatcher/releases/tag/v2.6.3) for now.
I will reverse changes which break the patch on some devices.
Issues with galaxy watch and galaxy buds needs more modification. Look at the [Readme](https://github.com/3arthur6/BluetoothLibraryPatcher#galaxy-watch-devices-support)

XDZemd, 2023-09-30:
Thank you so much for your reply.
I've tried your suggestion and I got stunk on some steps when I tried to use those commands.
So I changed the official rom and the bluetooth works fine. 
Thanks again for your reply and your hard work.
Best wishes for you!

closed 2023-10-01 by 3arthur6