3arthur6/BluetoothLibraryPatcher #48: [not an issue] 'su' needs to be mounted on 'com.android.bluetooth'

coldified, 2023-10-09:
As the title says, su needs to be mounted on 'com.android.bluetooth' for Magisk forks that has  'sulist' feature such as Magisk Delta.

Would be great if you mentioned this on the README. Thanks.

3arthur6, 2023-10-09:
Thanks
I wasn't aware of this and can't find documentation about it
Do you know if there is some shell command to deal with this sulist?

coldified, 2023-10-09:
> Thanks I wasn't aware of this and can't find documentation about it Do you know if there is some shell command to deal with this sulist?

Never made a magisk module, but:
```shell
# magiskhide
MagiskHide Config CLI
Usage: magiskhide [action [arguments...] ]
Actions:
   status          Return the MagiskHide status
   enable          Enable MagiskHide
   disable         Disable MagiskHide
   add PKG [PROC]  Add a new target to the hidelist (sulist)
   rm PKG [PROC]   Remove target(s) from the hidelist (sulist)
   ls              Print the current hidelist (sulist)
   exec CMDs...    Execute commands in isolated mount
                   namespace and do all unmounts
Magisk Delta specific Actions:
   sulist          Return the SuList status
   sulist [enable|disable]
                   Enable or disable SuList (need reboot)
   version         Print MagiskHide version
   --do-unmount [PID...]
                   Unmount all Magisk modifications
                   directly [in another namespace...]
# magiskhide sulist
SuList is enforced
# magiskhide add com.android.bluetooth
Target already exists in hidelist
```

Not sure if a magisk module can access magiskcli, but hope this helps.

3arthur6, 2023-10-09:
Sure it will!
Thanks, this will be fixed on next release with [21d47c9](https://github.com/3arthur6/BluetoothLibraryPatcher/commit/21d47c9efb04f23846f735dcb87265bf4ca335fd) commit