3arthur6/BluetoothLibraryPatcher #49: Dropping bluetooth connections after reboot

gdw1986, 2023-10-18:
Hi!
I'm using fold4, OneUI5.1.1, rooted, Patcher 2.6.5 has been intalled in Magisk26.3(not delta). The installation of seems alright:
- Copying zip to temp directory
- Installing BluetoothLibraryPatcher_module_2.6.5.zip
- Device is system-as-root
Archive:  /data/user/0/com.soul.mm/cache/flash/install.zip
  inflating: module.prop
***************************
 Bluetooth Library Patcher 
 by 3arthur6 
***************************
*******************
 Powered by Magisk 
*******************
Archive:  /data/user/0/com.soul.mm/cache/flash/install.zip
  inflating: customize.sh
- Extracting module files
Archive:  /data/user/0/com.soul.mm/cache/flash/install.zip
  inflating: 7z.tar.xz
  inflating: README.md
  inflating: SUPPORT.md
  inflating: bash.tar.xz
  inflating: busybox.tar.xz
  inflating: customize.sh
  inflating: hexpatch.sh
  inflating: latestchange.log
  inflating: module.prop
  inflating: service.sh
  inflating: update.json
- Magisk Manager installation
- Searching for relevant hex byte sequence
- Applying patch
- Successfully patched!
- Creating OTA survival service
- Done

But the Bluetooth connections will be dropped after rebooting or switching airplane mode on/off. Any suggestion?

gdw1986, 2023-10-20:
I think I found the cause of the problem, the module does not work in shamiko whitelist mode

cinzas, 2023-10-22:
Hi

I am having the same issue.
What have you done, removed shamiko ?

I need shamiko to hide Company Portal.

gdw1986, 2023-10-25:
> Hi
> 
> I am having the same issue. What have you done, removed shamiko ?
> 
> I need shamiko to hide Company Portal.

just turn off the white list mode

coldified, 2023-10-29:
> I think I found the cause of the problem, the module does not work in shamiko whitelist mode

You might need to add **com.android.bluetooth** to the Shamiko whitelist.

cinzas, 2023-10-30:
@coldified : Thanks for the input. Adding to the denylist solved the issue.
@gdw1986 : If I turn off the denylist, google pay, bank apps, company portal won't work, that is not the goal.

3arthur6, 2024-02-08:
Can this issue be considered solved?

closed 2024-02-08 by 3arthur6

grasmanek94, 2024-04-08:
One tip: Maybe add this to the docs, to not add com.android.bluetooth to Magisk DenyList (that was my issue).

IcyColdified, 2024-04-08:
> One tip: Maybe add this to the docs, to not add com.android.bluetooth to Magisk DenyList (that was my issue).

You are supposed to add it only if you have SuList or Whitelist mode enabled.