3arthur6/BluetoothLibraryPatcher #50: Only for non aisp roms Although on oneui 5.1

elswerky, 2023-11-23:
Hello
I used to flash this module on Note 10 lite
But I formatted device and flashed latest firmware October security patch 

And now when I flash it on Magick alpha it fails saying can't be use on asop rows 

Although am one oneuj v5.1 

Also tri3d to flash twep file but it stops at patching libraries

fragtion, 2023-12-30:
Same problem on SM910F with latest Magisk Alpha. looking forward to a fix

astrophilehuwan, 2024-01-06:
Same problem on Samsung A24, OneUI 6.0 and Magisk 26.4

privacyguy123, 2024-01-14:
Same problem - my fault for opening a duplicate issue.

Perhaps you could take a look? :)

Pharaoh2k, 2024-01-31:
@3arthur6 any news?

3arthur6, 2024-02-08:
Sorry for the delay
Latest [v2.6.6](https://github.com/3arthur6/BluetoothLibraryPatcher/releases/tag/v2.6.6) release should fix this issue.

shogunbr, 2024-02-08:
> Sorry for the delay Latest [v2.6.6](https://github.com/3arthur6/BluetoothLibraryPatcher/releases/tag/v2.6.6) release should fix this issue.

The Kitsune Mask variant of Magisk (which replaces Magisk Delta) has the same problem on installation. Could you please also fix it for Kitsune?

3arthur6, 2024-02-09:
> The Kitsune Mask variant of Magisk (which replaces Magisk Delta) has the same problem on installation. Could you please also fix it for Kitsune?

Just done it.

IcyColdified, 2024-02-10:
v2.6.8 now working on Galaxy A24 4G, OneUI 6.0, R65C33E4F-kitsune :+1:

3arthur6, 2024-10-06:
I assume this is now fixed this latest 2.7.0 release.

closed 2024-10-06 by 3arthur6