3arthur6/BluetoothLibraryPatcher #51: Please tell me what the problem is, I don't know what went wrong

Sakura5416, 2023-12-06:
In OneUI6.0, using the official Magisk module with Bluetooth Library Patcher can take effect, but after changing to Kitsune Mask, the Bluetooth will lose pairing after the phone restarts，Do I have to turn on sulist and put com.android.Bluetooth into sulist?

coldified, 2023-12-15:
If you already have SuList, you'll have to add `com.android.bluetooth` to it. If you're haven't enabled SuList, the module should work without further configuration.

Would be helpful if you provided more infos, do you have any other modules installed?

3arthur6, 2024-02-08:
> If you already have SuList enabled, you'll have to add `com.android.bluetooth` to it. If you haven't enabled SuList, the module should work without further configuration.
> 
> Would be helpful if you provided more infos, do you have any other modules installed?

So `com.android.bluetooth` isn't automatically added yet if SuList is enabled?

boudywho, 2024-04-15:
> If you already have SuList enabled, you'll have to add `com.android.bluetooth` to it. If you haven't enabled SuList, the module should work without further configuration.
> 
> Would be helpful if you provided more infos, do you have any other modules installed?

thank you so much, I was confused why it stopped working after moving to kitsune