3arthur6/BluetoothLibraryPatcher #55: [S24+] [A14] - Failed to patch

remdu92, 2024-02-06:
Hello,

I've tried to patch on a fresh new Samsung Galaxy S24+ (exynos) the Bluetooth Library without success.

Here the attached file
https://www.swisstransfer.com/d/7ad01b31-0492-47a4-8ae7-270c2ba0ed91

May you help ?
Thanks,
Best regards,

remdu92, 2024-02-08:
[magisk_install_log_2024-02-08T12.59.04.log](https://github.com/3arthur6/BluetoothLibraryPatcher/files/14208728/magisk_install_log_2024-02-08T12.59.04.log)

Hello, please find attached.

Best

3arthur6, 2024-02-08:
Thanks!
Could you try the attached zip?
[Bluetooth_Library_Patcher-v2.6.6_test.zip](https://github.com/3arthur6/BluetoothLibraryPatcher/files/14209267/Bluetooth_Library_Patcher-v2.6.6_test.zip)

remdu92, 2024-02-08:
[magisk_install_log_2024-02-08T15.29.44.log](https://github.com/3arthur6/BluetoothLibraryPatcher/files/14210486/magisk_install_log_2024-02-08T15.29.44.log)

Hello, please find a new log with the test file

3arthur6, 2024-02-09:
Huh
And what about this one?
[BluetoothLibraryPatcher_module_2.6.7_test.zip](https://github.com/3arthur6/BluetoothLibraryPatcher/files/14220974/BluetoothLibraryPatcher_module_2.6.7_test.zip)

remdu92, 2024-02-09:
this one is working, thanks a lot !

3arthur6, 2024-02-09:
thanks for the feedback, I will add the needed changes in the official release.

closed 2024-02-09 by 3arthur6