3arthur6/BluetoothLibraryPatcher #57: Smasung Galaxt A73

Parylo-D, 2024-02-10:
The module does not work on Galaxy A73 5G, Makisk Kitsune. Devices do not connect to the phone after turning on bluetooth

3arthur6, 2024-02-15:
Did you enabled SuList?
Try to add `com.android.bluetooth` to it.
With this command `magiskhide add com.android.bluetooth`

MrMasterDev, 2024-02-15:
> Did you enabled SuList? Try to add `com.android.bluetooth` to it. With this command `magiskhide add com.android.bluetooth`

This has been done for a long time, but it does not help. 
[image]
[image]

IcyColdified, 2024-02-16:
> Devices that are in the connected list do not reconnect to the phone after turning bluetooth on and off or rebooting the phone. 

Do you mean the devices are still on the "paired devices" list but not connecting to your device anymore? 
Or does the device completely disappear from the list after reboot?

MrMasterDev, 2024-02-16:
> Do you mean the devices are still on the "paired devices" list but not connecting to your device anymore? Or does the device completely disappear from the list after reboot?

The devices are still on the "paired devices" list but not connecting to my device anymore?

3arthur6, 2024-02-18:
Ok so it seems to be another problem. Devices will totally disappear of the list without this patch.
Which kind of devices do you use?
Try to uninstall the module when reboot. Then install it again and reboot again.
In last resort you can try to reset network settings.

MrMasterDev, 2024-02-19:
> > The devices are still on the "paired devices" list but not connecting to my device anymore.
> 
> Ok so it seems to be another problem. Devices will totally disappear of the list without this patch. Which kind of devices do you use? Try to uninstall the module when reboot. Then install it again and reboot again. In last resort you can try to reset network settings.

I have already done all this, several times and it did not help. Now I only use bluetooth headphones, sometimes I connect them to a TV or a speaker.

3arthur6, 2024-02-20:
Huh.. Maybe another module.
Have you tried disabling all other modules?

MrMasterDev, 2024-02-20:
> Huh.. Maybe another module. Have you tried disabling all other modules?

What are the other modules for?

3arthur6, 2024-02-20:
Magisk modules.
Do you have any other ones?

MrMasterDev, 2024-02-20:
> Magisk modules. Do you have any other ones?

There are other modules, but what do they have to do with your module?

IcyColdified, 2024-02-21:
> There are other modules, but what do they have to do with your module?
> Isn't this a compatibility issue with magisk kitsune? 

It is indeed unlikely for other magisk modules to conflict with this module, but chances aren't zero xd

> I turned off all the modules, but it didn't help. But I found that everything works, but I constantly have to turn off and turn on the headphones. 

Does the exact same thing happen to the other devices you have mentioned?

IcyColdified, 2024-02-23:
> > > There are other modules, but what do they have to do with your module?
> > > Isn't this a compatibility issue with magisk kitsune?
> > 
> > It is indeed unlikely for other magisk modules to conflict with this module, but chances aren't zero xd
> > 
> > > I turned off all the modules, but it didn't help. But I found that everything works, but I constantly have to turn off and turn on the headphones.
> > 
> > Does the exact same thing happen to the other devices you have mentioned?
> 
> So the same thing happens with other devices 

What have you done so far to investigate / resolve this issue? (Sorry)

MrMasterDev, 2024-02-25:
> What have you done so far to investigate / resolve this issue? (Sorry)
I did everything possible. I followed all your recommendations. I don't know what else can be done.

MrMasterDev, 2024-03-22:
> What have you done so far to investigate / resolve this issue? (Sorry)

After long tests, I found that in fact, it automatically connects to devices after restarting the phone, if the device to which you connect is turned off and then turned on, it is also automatically connected to it. Only if you turn off bluetooth on the phone and then turn it on, it will not automatically connect to the device to which it is connected.

3arthur6, 2024-03-22:
> After long tests, I found that in fact, it automatically connects to devices after restarting the phone, if the device to which you connect is turned off and then turned on, it is also automatically [...]

So you don't loose your paired devices right?
This issue is related to auto connect?
The module doesn't change anything to this.

MrMasterDev, 2024-03-22:
> So you don't loose your paired devices right? This issue is related to auto connect? The module doesn't change anything to this.
 
Yes.I don't loose my paired devices. This issue is related to auto connect.

3arthur6, 2024-03-22:
> Yes.I don't loose my paired devices. This issue is related to auto connect.Is there any way to fix this? 

This module doesn't deal with auto connect sorry.
I don't have any information about this.

closed 2024-03-22 by MrMasterDev