3arthur6/BluetoothLibraryPatcher #59: the module thinks that I don’t have a Samsung

HIB1tch123, 2024-03-25:
[image]

Bluetooth was turned off at the time of installation

maybe one of these modules is interfering with the installation 
[image]
[image]

IcyColdified, 2024-03-25:
Are you using vanilla Magisk or others (Alpha, Kitsune/Delta, etc)?

Most modules shouldn't affect the installation of the module unless they heavily modify the build.prop, I believe.

HIB1tch123, 2024-03-25:
I use regular magisk 27.0

IcyColdified, 2024-03-26:
Have you tried disabling all the other modules, rebooting, and then installing the module?

HIB1tch123, 2024-03-26:
this helped thanks, I really don’t understand why it didn't work before I also turned off all the modules and installed it on a completely clean system(

closed 2024-03-26 by HIB1tch123