3arthur6/BluetoothLibraryPatcher #65: Patcher not executing

paratkd, 2024-08-10:
I have a Galaxy S10 5G running stock Samsung firmware, albeit Magisk Kitsune patched for root.

Initially, the patch module refused to run as it was stating it is only for stock Samsung ROM; I skipped this by editing the customise.sh script.

However, now it gets to the "searching for hex byte pattern" stage and never moves any further. I let it run for a long time, no changes seem to be made. I also tried older version releases, but to no avail.

I'm currently stuck with my phone not pairing properly to my Wearable Watch6 (although buds seems to be fine.)

Many thanks 👍

usaingoldx, 2024-09-23:
Same device, same error. (I do close magisk app because is stuck same text. After i do reboot device , login magisk app and i see bluetooth library patcher module in the modules page. But not look working.)

3arthur6, 2024-10-06:
Could anyone confirm 2.7.0 fixed this too?

closed 2024-10-09 by 3arthur6