3arthur6/BluetoothLibraryPatcher #67: S24 Plus OneUi 6.1->6.1.1 update, bluetooth crashing

KrazyDeinos, 2024-09-17:
Updated to OneUi 6.1.1 and after update Bluetooth is crashing.
Tested to disable reboot enable the module, no effect. 

A re-install failed with the error message like other users have reported:
"only for samsung stock roms not relevant for aosp roms!"

Tested disabling the modules Tricky Store and Play Integrity Fork, no effect.

6.1.1 was a big update for OneUi, so this module may need some tweaking for it to work.

KrazyDeinos, 2024-09-18:
Log

`- Copying zip to temp directory
- Installing BluetoothLibraryPatcher_module_2.6.9.zip
- Device is system-as-root
Archive:  /data/user/0/com.topjohnwu.magisk/cache/flash/install.zip
  inflating: module.prop
***************************
 Bluetooth Library Patcher 
 by 3arthur6 
***************************
*******************
 Powered by Magisk 
*******************
Archive:  /data/user/0/com.topjohnwu.magisk/cache/flash/install.zip
  inflating: customize.sh
- Extracting module files
Archive:  /data/user/0/com.topjohnwu.magisk/cache/flash/install.zip
  inflating: 7z.tar.xz
  inflating: 7z64.tar.xz
  inflating: README.md
  inflating: SUPPORT.md
  inflating: bash.tar.xz
  inflating: bash64.tar.xz
  inflating: busybox.tar.xz
  inflating: customize.sh
  inflating: hexpatch.sh
  inflating: latestchange.log
  inflating: module.prop
  inflating: service.sh
  inflating: update.json
- Magisk Manager installation
grep: /debug_ramdisk/.magisk/mirror/system/build.prop: No such file or directory
- Only for Samsung stock based roms!
- Not relevant for aosp roms!

! Installation failed
`

KrazyDeinos, 2024-09-18:
According to this commit, from magisk 27005 an onwards, magisk isnt using mirror:
https://github.com/topjohnwu/Magisk/commit/92b305a389bcb0dc7e2b3169134f0a81bac6c513

Can I edit service.sh to this? :
`check() {
  samsung=`grep -Eqw "androidboot.odin_download|androidboot.warranty_bit|sec_debug" /proc/cmdline && echo 'true' || echo 'false'`
  if [[ ! -z $KSU_VER ]] ; then
    ui_print "- KernelSU Manager installation"
    sys=/system
  elif $BOOTMODE ; then
    ui_print "- Magisk Manager installation"
    if [[ $MAGISK_VER == *-alpha ]] || [[ $MAGISK_VER == *-kitsune ]]; then
      ui_print "- Magisk Alpha/Kitsune fork detected"
      sys=/system
    else
      # Use the system path directly since the mirror is no longer available
      sys=/system
    fi
  else`

3arthur6, 2024-09-29:
Thanks for pointing out this I will push an update right now.

KrazyDeinos, 2024-09-29:
> Thanks for pointing out this I will push an update right now.

@3arthur6  I tested 2.7.0 now , sadly 

`- Copying zip to temp directory
- Installing BluetoothLibraryPatcher_module_2.7.0.zip
- Device is system-as-root
Archive:  /data/user/0/com.topjohnwu.magisk/cache/flash/install.zip
  inflating: module.prop
***************************
 Bluetooth Library Patcher 
 by 3arthur6 
***************************
*******************
 Powered by Magisk 
*******************
Archive:  /data/user/0/com.topjohnwu.magisk/cache/flash/install.zip
  inflating: customize.sh
- Extracting module files
Archive:  /data/user/0/com.topjohnwu.magisk/cache/flash/install.zip
  inflating: 7z.tar.xz
  inflating: 7z64.tar.xz
  inflating: README.md
  inflating: SUPPORT.md
  inflating: bash.tar.xz
  inflating: bash64.tar.xz
  inflating: busybox.tar.xz
  inflating: customize.sh
  inflating: hexpatch.sh
  inflating: latestchange.log
  inflating: module.prop
  inflating: service.sh
  inflating: update.json
sh: 0495468d: bad number
- Magisk Manager installation
grep: /debug_ramdisk/.magisk/mirror/system/build.prop: No such file or directory
- Only for Samsung stock based roms!
- Not relevant for aosp roms!

! Installation failed
`
Perhaps the 6.1.1 changed quite a bit from 6.1

3arthur6, 2024-09-29:
Thanks, I pushed the update a bit to quick..
It should be ok.

KrazyDeinos, 2024-09-29:
> Thanks, I pushed the update a bit to quick.. It should be ok.

Thanks @3arthur6 , now it goes further, although stuck at "searching for relevant hex byte sequence" , and later "applying patch"
I saw this in the last release, with and older magisk, to see if the module would work with the 6.1.1 update. 
I had it be stuck there for about  30min before i rebooted, and after  a reboot with magisk module install still stuck, the module would be active but not working, and bluetooth crashing as an result.

But maybe it will be different this time and "applying patch" finishes.

3arthur6, 2024-09-29:
ok!
Could you please attach here the file /system/apex/com.android.btservices.apex from you phone?
You will need a root file manager.

KrazyDeinos, 2024-09-29:
> ok! Could you please attach here the file /system/apex/com.android.btservices.apex from you phone? You will need a root file manager.

Here you of @3arthur6 , it's zipped because of GitHub attachment limitations 
[Uploading com.android.btservices.zip…]()

3arthur6, 2024-09-29:
I think you sent the message before the file was fully uploaded I can't download it

KrazyDeinos, 2024-09-29:
[com.android.btservices.zip](https://github.com/user-attachments/files/17180983/com.android.btservices.zip)
@3arthur6  sorry buddy =)

3arthur6, 2024-09-29:
Thanks dude! It will need more investigation on my part

KrazyDeinos, 2024-09-29:
Sure thing! Happy to help, and good hunting!
You've been awesome pushing this module out for users to experience bluetooth to work as it should work =)

KrazyDeinos, 2024-09-29:
@3arthur6 the install actually finished, after roughly an hour, and the module is working, did a few reboot tests and connection tests. All good!
Maybe this will be somewhat helpful 😀

KrazyDeinos, 2024-10-03:
> @KrazyDeinos Could you try with one? [BluetoothLibraryPatcher_module_2.7.0.zip](https://github.com/user-attachments/files/17248729/BluetoothLibraryPatcher_module_2.7.0.zip) I should fix performance regression and so installation time.

@3arthur6 you are a magician! Tested now, instant install, bt devices still remembered, Bluetooth connects and running fine ! 🍺😀

3arthur6, 2024-10-03:
Thanks dude!
I will release it soon

3arthur6, 2024-10-06:
I assume this is now fixed. Feel free to open again if any issue occurs.

closed 2024-10-06 by 3arthur6