3arthur6/BluetoothLibraryPatcher #71: The 2.7.1 version doesn't work on Samsung galaxy a54 5g  with one ui 6.1 even if it patches

angaratosurion, 2024-10-28:


kigung9029, 2024-11-20:
Possibly because of Tricky Store or TSupport Advance module, try to remove all of those module first.

angaratosurion, 2024-11-20:
I dont have tricky store module?
Also remove them before patching and re add the later?

kigung9029, 2024-11-21:
Sorry, I got it wrong. After more testing the bluetooth lib patcher doesn't work when using Shamiko in whitelist mode. If you still can't get it to work, I recommend you to try remove other modules first.

angaratosurion, 2024-11-22:
i don' have shamiko , i stopped using it months before

angaratosurion, 2024-11-24:
[image]
[image]

[image]

These are the modules I use