3arthur6/BluetoothLibraryPatcher #73: Install seems to hang on applying patch on s24u beta (a15)

exocetdj, 2024-12-05:
Install hangs at applying patch on a15 oneui 7 beta. Please see attached install logs
[magisk_install_log_2024-12-05T17.46.05.log](https://github.com/user-attachments/files/18030245/magisk_install_log_2024-12-05T17.46.05.log)
[magisk_install_log_2024-12-05T17.40.20.log](https://github.com/user-attachments/files/18030246/magisk_install_log_2024-12-05T17.40.20.log)

3arthur6, 2024-12-07:
[BluetoothLibraryPatcher_module_2.7.2.zip](https://github.com/user-attachments/files/18047954/BluetoothLibraryPatcher_module_2.7.2.zip)

Is this one working for you?

3arthur6, 2024-12-07:
> Hi, yep this one seems to be working fine 🙂 👍 I can see the necessary files have now installed in /data/adb/modules

Sure you keep bluetooth paired devices after reboot? I got some bad feedback from others users

exocetdj, 2024-12-07:
OK done some further testing and for some reason Bluetooth connection is
remembered for buds and my Bluetooth adapter (non Samsung brand) WITHOUT
any patch installed so any positive result seems to be made moot as Samsung
seem to have left Bluetooth alone on the beta!

I think Dr ketan also found this with Samsung buds

closed 2025-07-26 by 3arthur6