3arthur6/BluetoothLibraryPatcher #74: S24_OneUI7.0 update bluetooth patch

jrkruse, 2024-12-07:
needs updated for latest hex patch needs updated
[BluetoothLibPatcher-files_2.zip](https://github.com/user-attachments/files/18046017/BluetoothLibPatcher-files_2.zip)

3arthur6, 2024-12-07:
[BluetoothLibraryPatcher_module_2.7.2.zip](https://github.com/user-attachments/files/18047954/BluetoothLibraryPatcher_module_2.7.2.zip)

Is this one working for you?

J0SH1X, 2024-12-07:
This one flashes but doesn't work after reboot my bt devices are still gone on s23 one ui 7

3arthur6, 2024-12-07:
It sounds bad as the files provided show some big compared to the last four one ui updates...

3arthur6, 2024-12-07:
> This one flashes but doesn't work after reboot my bt devices are still gone on s23 one ui 7

Could you upload here your /data/misc/bluedroiddump folder?

J0SH1X, 2024-12-07:
Sure give me 10 minutes

J0SH1X, 2024-12-07:
> /data/misc/bluedroiddump

the directory is empty

3arthur6, 2024-12-07:
> > /data/misc/bluedroiddump
> 
> the directory is empty

Damn.. it will be more hard so. Could you upload the magisk module install log?

J0SH1X, 2024-12-07:
where would that be located

3arthur6, 2024-12-07:
> where would that be located

Install the module again and at the end you have an icon in the top right to save the installation log.

> [subBuffer.log](https://github.com/user-attachments/files/18048333/subBuffer.log) i got the log

It looks like the module is not installed or doesn't have any effect..

J0SH1X, 2024-12-07:
> Install the module again and at the end you have an icon in the top right to save the installation log.
>
>
> It looks like the module is not installed or doesn't have any effect..

[magisk_install_log_2024-12-07T13.51.18.log](https://github.com/user-attachments/files/18048383/magisk_install_log_2024-12-07T13.51.18.log)

3arthur6, 2024-12-07:
> 6765744B657949564C656E203D207B7D 6765744B657949564C656E203D203136
> 
> From: "getKeyIVLen = {}" To: "getKeyIVLen = 16"
> 
> This should fix the issue

Wow sure? Did you try it?

3arthur6, 2024-12-07:
[BluetoothLibraryPatcher_module_2.7.2.zip](https://github.com/user-attachments/files/18049112/BluetoothLibraryPatcher_module_2.7.2.zip)

This one follows @bklynali patch

J0SH1X, 2024-12-07:
Testing

J0SH1X, 2024-12-07:
Meh can't upload logs from phone.
Doesn't work still

jrkruse, 2024-12-07:
> [BluetoothLibraryPatcher_module_2.7.2.zip](https://github.com/user-attachments/files/18049112/BluetoothLibraryPatcher_module_2.7.2.zip)
> 
> This one follows @bklynali patch

this worked for me

3arthur6, 2024-12-07:
@J0SH1X you sure it doesnt work? As I get Install the module be sure to reboot then try pairing and reboot.

> > Meh can't upload logs
> > from phone. Doesn't work still
> 
> Try this
> 
> [BluetoothLibraryPatcher_module_2.7.2A.zip](https://github.com/user-attachments/files/18049934/BluetoothLibraryPatcher_module_2.7.2A.zip)
> 
> based on the log I added few more

Hey only hex string 135 is applied with the patch. It relies to the android api, 35 with one ui 7.

J0SH1X, 2024-12-08:
Yeah, i think it may be a port rom iusse cause it still doesnt work despite seeing reports of it working on s24

jrkruse, 2024-12-08:
The Bluetooth issue is very odd sometimes it works  and remember pairings with no patch needed. Other times you install and it doesn’t remember pairings and needs the patch module. You got me

J0SH1X, 2024-12-08:
[subBuffer.log](https://github.com/user-attachments/files/18053997/subBuffer.log)
@bklynali

jrkruse, 2024-12-09:
I didn’t mean to close issue sorry

3arthur6, 2024-12-10:
Wow it never needed as much patches..
What did you changes actually?

3arthur6, 2024-12-10:
Wow it never needed as much patches.
What did you changes actually?

Mark8980, 2025-02-17:
same issue on S25ultra. BT connection working with buds 3 pro though it isnt working with galaxy watch 5. connection lost after rebooting the phone

3arthur6, 2025-02-18:
I will take a look on the official OneUI7 firmware

3arthur6, 2025-02-19:
@Mark8980 
Is this one fixing the issue?

[BluetoothLibraryPatcher_module_2.7.3.zip](https://github.com/user-attachments/files/18876136/BluetoothLibraryPatcher_module_2.7.3.zip)

Mark8980, 2025-02-19:
yes this one is working for me. thanks a lot