3arthur6/BluetoothLibraryPatcher #89: S24 Ultra A16 one UI 8.0 hangs on installing

jrkruse, 2025-07-20:
S24 ultra hangs on applying patch it works on s23 but not on s24

[magisk_install_log_2025-07-19T23.28.35.log](https://github.com/user-attachments/files/21333007/magisk_install_log_2025-07-19T23.28.35.log)

jrkruse, 2025-07-20:
Here is the lib file from apex file

[libbluetooth_jni.zip](https://github.com/user-attachments/files/21333034/libbluetooth_jni.zip)

jrkruse, 2025-07-20:
I flashed the module in twrp it failed but here is the tar file it created

[BluetoothLibPatcher-files.zip](https://github.com/user-attachments/files/21333162/BluetoothLibPatcher-files.zip)

jrkruse, 2025-07-25:
> Hey Thanks to your log and libbluetooth file I changed the patch to include your and probably more device! You can try latest 2.9.1 version

Seems to work now on what I tested it on.

3arthur6, 2025-07-26:
Thanks! I close it for now

closed 2025-07-26 by 3arthur6